sousandrei/ism330dhcx #6: ODR for accelerometer and gyro is in kHz not Hz

gauteh, 2021-12-14:
In:

```

/// Accelerometer ODR
///
/// Default value: `Off`
#[allow(non_camel_case_types)]
#[derive(Copy, Clone, Eq, PartialEq, Ord, PartialOrd, Hash, Debug)]
pub enum Odr_Xl {
    Off,   // off
    Hz125, // 12.5 Hz
    Hz26,  // 26   Hz
    Hz52,  // 52   Hz
    Hz104, // 104  Hz
    Hz208, // 208  Hz
    Hz416, // 416  Hz
    Hz833, // 833  Hz
    Hz166, // 1.66 Hz
    Hz333, // 3.33 Hz
    Hz666, // 6.66 Hz
}
```

the last three values should be in kHz. Same issue for gyroscope.

I don't think the actual functionality is affected. Some of the combinations are only valid depending on performance mode. That would probably only be enforced with some redesign and making register access less direct.

sousandrei, 2022-01-09:
do you think updating the docs would be sufficient for now? 
+1 on making the register access less direct, I need to also make a cleaner way of storing the address and the actual register value, haven't cracked that yet

gauteh, 2022-01-09:
I think it is fairly confusing with `Hz166` compared to `kHz166`, I would definitely try changing that as soon as possible. Maybe even use `kHz6_666`. Or even `Hz6660`. All of them less nice, but at least correct. I often don't realize the docs have different values in this case, since the enum seems so obvious to understand in the first place.

Some of the other crates linked in #1 has taken a somewhat higher level approach, then it is a bit easier to enforce the right usage. But it is a bit of work moving to that, so I don't expect that you want to spend time on that.

sousandrei, 2022-01-09:
Just pumped out a new version renaming them with the proper `khz` :D

gauteh, 2022-01-09:
Great!! Thanks

closed 2022-01-26 by gauteh